emoncms/emoncms #1776: PHP 8.1 issue

thib66, 2022-01-14:
Hello,
the migration from PHP 8.0 to 8.1 generates somme errors when trying to logon : 
[image]
[image]

In details : 
Deprecated: Constant FILTER_SANITIZE_STRING is deprecated in /www/emoncms/locale.php on line 130

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 113

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 144

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 144

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 113

Deprecated: parse_str(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/Modules/user/user_controller.php on line 35

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 113

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 113

Deprecated: Constant FILTER_SANITIZE_STRING is deprecated in /www/emoncms/Modules/user/user_controller.php on line 38

Deprecated: Constant FILTER_SANITIZE_STRING is deprecated in /www/emoncms/locale.php on line 130

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 144

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 144

Deprecated: stripslashes(): Passing null to parameter #1 ($string) of type string is deprecated in /www/emoncms/core.php on line 113
{"success":true,"message":"Connexion r\u00e9ussie","startingpage":"user\/view"}

thib66, 2022-01-14:
A solution is proposed here : https://github.com/emoncms/emoncms/pull/1777.

Nevetheless, I am not sure that htmlspecialchars() is the correct function to replace the deprecated constant FILTER_SANITIZE_STRING.

@glynhudson @TrystanLea , as the substitution of FILTER_SANITIZE_STRING with another non deprecated PHP function has an impact of the global security of the application. I would like to have your opinion.

Do we use htmlspecialchars() or do we use another function ?
Thanks in advance for your answer.

thib66, 2022-01-20:
There is also another issue when typing wrong credentials : 
[image]
A solution is proposed here : #1777.

thib66, 2022-01-31:
New issue met in vis module : 
[image]
fix also proposed in #1777

TrystanLea, 2023-04-06:
all fixed now I believe? closing for now

closed 2023-04-06 by TrystanLea