emoncms/emoncms #1778: Raspberry Pi install instructions not working for Buster

jsmarion, 2022-01-17:
Hi - Would it be possible to list the install steps for Buster? My RPi seems to be having trouble with the PHP packages. Many thanks. Output from the apt-get install step:

Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php7.0-thrift' for regex 'php7.0'
Note, selecting 'php7.0-remctl' for regex 'php7.0'
Note, selecting 'php7.0-common' for regex 'php7.0'
Note, selecting 'php7.0-curl' for regex 'php7.0'
Note, selecting 'php-remctl' instead of 'php7.0-remctl'
Package mysql-client is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  mariadb-client-10.0

Package php7.0-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package php7.0-curl is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'mysql-client' has no installation candidate
E: Unable to locate package libapache2-mod-php7.0
E: Couldn't find any package by glob 'libapache2-mod-php7.0'
E: Couldn't find any package by regex 'libapache2-mod-php7.0'
E: Unable to locate package php7.0-mysql
E: Couldn't find any package by glob 'php7.0-mysql'
E: Couldn't find any package by regex 'php7.0-mysql'
E: Unable to locate package php7.0-gd
E: Couldn't find any package by glob 'php7.0-gd'
E: Couldn't find any package by regex 'php7.0-gd'
E: Unable to locate package php7.0-opcache
E: Couldn't find any package by glob 'php7.0-opcache'
E: Couldn't find any package by regex 'php7.0-opcache'
E: Package 'php7.0-curl' has no installation candidate
E: Unable to locate package php7.0-dev
E: Couldn't find any package by glob 'php7.0-dev'
E: Couldn't find any package by regex 'php7.0-dev'
E: Unable to locate package php7.0-mcrypt
E: Couldn't find any package by glob 'php7.0-mcrypt'
E: Couldn't find any package by regex 'php7.0-mcrypt'
E: Package 'php7.0-common' has no installation candidate
E: Unable to locate package php7.0-mbstring
E: Couldn't find any package by glob 'php7.0-mbstring'
E: Couldn't find any package by regex 'php7.0-mbstring'

borpin, 2022-01-17:
> Would it be possible to list the install steps for Buster?

What instructions are you trying to follow?

jsmarion, 2022-01-17:
They are located here: https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/readme.md under the heading 'Archived: Install Emoncms on Raspberry Pi (Raspbian Stretch)'. I got there via a link from another project (https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter#setting-up-emoncms), but now I see that that link may be old and that there is a script. I suppose I should try that?

borpin, 2022-01-17:
Yes exactly *Archived* i.e. do not use - read the text above.

[image]

jsmarion, 2022-01-17:
Yeah. Should have done that already, but I appreciate the help. I am adding it to an existing RPi (running Pi-Hole) so working through the instructions on how to manually set up the data partition (modifying them for a larger SD card).

[image]<https://user-images.githubusercontent.com/1322749/149833566-6b10ed0c-50d9-4795-b899-e1b039227a29.png>

borpin, 2022-01-17:
> I am adding it to an existing RPi (running Pi-Hole) so working through the instructions on how to manually set up the data partition (modifying them for a larger SD card).

Generally a very bad idea. It will end in tears. Emoncms will happily run on a PiZero.

jsmarion, 2022-01-17:
I’d rather use a PiZero than cry. Thanks for the advice!

borpin, 2022-02-04:
@jsmarion - please close...

jsmarion, 2022-02-04:
Thanks for your help. Got it up and running now.

closed 2022-02-04 by jsmarion