emoncms/emoncms #1782: help ... probable stupid question but ...

maurofilippo, 2022-02-11:
hi everyone, i have been using emoncms 8.4.0 on a hosting service for a couple of years now and everything works great.
Now I have installed the latest version of Emoncms via sd on a Raspberry, here too everything is ok.
The problem is that I would like to backup, if possible, from the hosting to the local raspberry ... and I can't understand how .... I have read all the guides but I don't understand .... With filezilla from my pc windows I have backed up where emoncms resides. I put everything on a stick and, after connecting it to the raspberry, I tried from emoncms - backup, but it doesn't work, it writes too many partitions error ....

Can you help me, fool-proof steps, because I can't figure out if something is wrong or it can't be done ....

thank you very much for those who want to help me.
good evening
Mauro

alexandrecuer, 2022-02-11:
Did you try the sync module to transfer feed by feed ?

maurofilippo, 2022-02-12:
Thanks for the reply. No, I haven't tried ... but what I can't understand is if I have to do all these things from my raspberry (and here it would be difficult for me since I don't know it) or from my windows pc and then transfer to the raspberry with filizilla ? Unfortunately I lack the basics .... that's why I was asking for something stupid ... I would be sorry to lose the many feeds that over the years have recorded the data in my house, boiler, outside temperatures, humidity, home automation activities, etc. etc. Thanks again . ps I looked at what you recommended, I saw on a 2012 forum a php program to download to synchronize data ... is it correct?
 Hi everyone

alexandrecuer, 2022-02-12:
Sync should ne preinstalled on your pi if you used an emonpi sd image : 

[image]

[image]

maurofilippo, 2022-02-12:
here I am (sorry for my english but i am using a translator)

then, I tried as you wrote me, I clicked on "sync" I went to "sync feeds" I entered the host address (here the question ... I have to put the address where there is the file "index .php "or should I continue with the" /feed/list.php "folders?)
... but it doesn't work, it tells me that the credentials are wrong (it's not true) but also writes "who is downloading them ..." (?!?) .... I also tried to put the address without " / feed "and here he does not write me wrong credentials but does nothing else ...

Now I don't know if it takes that long (that's a lot of feeds and they've been recording for 2 years) and it's just stuck ... but it doesn't show anything even after 10 minutes ...

I tried to click "sync input" and here it worked! And he dumped all the inputs and loaded me into the raspberry (GREAT !!)

With dashboard, it gave me a series of errors, but still it created the 5 dashboards but white (but this is not a problem, I can make them myself ...)

The feeds problem remains ... can you give me a hand again please .... I don't know what to do ....

PS the emoncms folder that runs on the host I downloaded it to the hard disk of the windows computer if it can be of help to let me do some tests ....

thank you so much again.
Hello
Mauro

[image]
[image]
[image]
[image]
[image]

maurofilippo, 2022-02-12:
this is the photo of when I click "sync dashboard", "download" appears and as soon as I click it writes me those errors, creating the white dashboards (but I repeat this for me is the least of the problems .... I'm interested in the data feeds)

[image]

alexandrecuer, 2022-02-13:
Your emoncms version on the server seems quite old but it should be ok

On the pi, go to the admin module and check the sync branch. If you are on the master branch of emoncms, you should be on the master branch of sync. After that try again to synchronise the feeds between the 2 machines

maurofilippo, 2022-02-13:
hello, sorry but I didn't understand where I have to go ...
On the raspberry (where I have the new emoncms program) I went into administration but I can't find sync ... or I don't see it ... I made you screenshots of the whole screen.
I also made you the admin photo emoncms 8.4.0 (the host version) which runs in the hosting service

Thanks again

[image]
[image]
[image]
[image]
[image]

maurofilippo, 2022-02-15:
Hello. then I went to "administration" of the raspberry and then into "components" and you see the photos I took ... I pressed update on "sync" and those logs came out at the top of the page ... both as "master" that as "stable" ...
then I went to sync "not in administration" and I put the login credentials back but, as usual, on feeds I find the blank page ...

i attached the logs to you (i deleted apikey write and read ... which are correct, they are the same as the version in host emoncms 8.4.0)

I still think I didn't understand what you told me to do ... sorry ... maybe it's the translator or .... me ;-)

[image]
[image]
[image]
[image]

maurofilippo, 2022-02-17:
Hello! I have a new update even if I think it doesn't change my situation much ... I installed the latest version of emoncms in another hosting service, I created feeds and also the dashboard ... then I connected with the raspberry and I did the synchronization and everything was ok! For me a certainty that the raspberry does what it should do! Now, I can make some changes to aim to download version 8.4.0 feeds .... I hope so .... bye and have a good day Mauro

borpin, 2022-02-17:
These things are not issues - ask for help on the forum https://community.openenergymonitor.org/

maurofilippo, 2022-02-18:
Ok, sorry. I'll try on the forum.
 thank you 
Mauro