emoncms/emoncms #1784: /feed/insert.json no longer updates last time and value in 'feeds' table [MySQL]

MrTimbones, 2022-02-25:
Having just updated from version 10.8.5 to 11.0.5, I noticed that the feeds table is no longer being updated when updating feeds directly via the Feed API. [Redis not in use here, only MySQL]

Using /feed/insert.json to bulk upload some history for a handful of feeds, I can see the data being correctly stored in EmonCMS, yet the last time and last value for those feeds is stuck at just before I upgraded.

TrystanLea, 2022-02-25:
Thanks for spotting @MrTimbones I've updated the master branch with a fix for this, would you be able to test there? Il get it merged to stable soon. Cheers.

MrTimbones, 2022-02-25:
Yes, that seems to have done the trick. Thanks for fixing so quick!