emoncms/emoncms #1788: dashboard: kwhperiod error -> kwhperiod_render.js?v=3 -> error: result is null

camelcamro, 2022-03-11:
emoncms:
-> version: 
Git
    URL
        https://github.com/emoncms/emoncms.git
    Branch
        * master
    Describe
        11.0.5-3-g122d3291

Components
    Emoncms Core v11.0.7 | App v2.5.1 | Dashboard v2.2.2 | Device v2.1.7 | Graph v2.1.6 | WiFi v2.1.1 | Backup v2.3.2 | DemandShaper v2.2.2 | Postprocess v2.2.4 | Sync v2.1.3 | Emonhub v2.3.2 | EmonPi v2.9.5 | EmonScripts v1.4.1 | RFM2Pi v1.4.1

#############################

i'm using the khwperiod since around 1-2 years, without any problems, and without doing any changes in dashboard to show the daily usage of few values.
and seems to be, auto-updates are executed and upgraded or don'T know.
but, now it is not working around since 2 weeks (~2022-02-22)

i already tried to cleanup browsers and so on ...cache cleanup/reload, then tested with fresh new windows version with mozilla + chrome -> still getting this error:

error when using 
EmonCMS Error
-------------
Message: TypeError: result is null
Route: Modules/dashboard/widget/kwhperiod/kwhperiod_render.js?v=3
Line: 746
Column: 34

MrTimbones, 2022-03-16:
The bug is here is that `feed.getvalue()` returns only the value, but the line that follows treats it as an array of `[time, value]`.

Removing `[1]` from line 746 fixes the error reported above, and appears to produce the current numbers for me.

TrystanLea, 2022-03-16:
Thanks @MrTimbones that's great, I've applied the fix https://github.com/emoncms/dashboard/commit/9b5756ef6275b9c36cd3ba70509da74ff31ff0da thanks again

closed 2022-03-16 by TrystanLea