emoncms/emoncms #1791: Can't get last data point from MySQL TimeSeries using Daily/Weekly/Yearly aggregation [patch]

MrTimbones, 2022-03-20:
I have some daily feeds that are only updated once a day. I noticed that I could never get the graph to show data for yesterday.

I've tracked this down to [MysqlTimeSeries.php](https://github.com/emoncms/emoncms/blob/master/Modules/feed/engine/MysqlTimeSeries.php) which could do with a couple tweaks:
```diff
diff --git a/Modules/feed/engine/MysqlTimeSeries.php b/Modules/feed/engine/MysqlTimeSeries.php
index 5bdc160d..808122e2 100644
--- a/Modules/feed/engine/MysqlTimeSeries.php
+++ b/Modules/feed/engine/MysqlTimeSeries.php
@@ -438,7 +438,7 @@ class MysqlTimeSeries implements engine_methods
                 }
             } else {
                 // Limit DB requests to available datapoints in feed
-                if ($start_time < $time && $time < $end_time) {
+                if ($start_time < $time && $time <= $end_time) {
                     // get datapoint using interpolation if necessary
                     $dp = $this->get_datapoint_interpolated($feedid, $time);
                     $value = $dp[1];
@@ -887,6 +887,9 @@ class MysqlTimeSeries implements engine_methods
                     }
                 }
             }
+            else if (count($dp) == 1 && $dp[0]['time'] == $time) {
+                $data = array($time, (float) $dp[0]['data']);
+            }
             else {
                 // only one datapoint found, interpolation not possible.
                 $data = array($time , null);
```
These changes ensure that `get_data_DMY()` requests all points _including_ the last one and `get_datapoint_interpolated()` will return the final point without interpolation. I now get all the days up to yesterday, and the data is the same as using Fixed Interval 86400.

I don't know if this bug also exists in other feed engines.

TrystanLea, 2022-03-25:
Thanks @MrTimbones I will take a look at this soon, really appreciate the diff on what fixed it for you that's really helpful!

TrystanLea, 2026-10-05:
Thanks @MrTimbones fixed in https://github.com/emoncms/emoncms/commit/b2d7e393f90dab451e6b650d9a6e555a56c9c7c4 sorry that this took so long!

closed 2026-10-05 by TrystanLea