emoncms/emoncms #1803: current emoncms master seems to fail on php8.1

warpme, 2022-05-23:
Hi,

I'm using emoncms on Archlinux. I worked well. 
Lastly i upgraded Archlinux to current ver. 
It has php8.1 and it looks emoncms is not ready for pho8.1.
Error reported by emoncms:

```
Fatal error: Uncaught TypeError: fclose(): Argument #1 ($stream) must be of type resource, bool given in /var/lib/emoncms/www/Lib/EmonLogger.php:43 Stack trace: #0 /var/lib/emoncms/www/Lib/EmonLogger.php(43): fclose() #1 /var/lib/emoncms/www/index.php(29): EmonLogger->__construct() #2 {main} thrown in /var/lib/emoncms/www/Lib/EmonLogger.php on line 43
```

borpin, 2022-05-24:
https://github.com/emoncms/emoncms/blob/d042db8924ce712111cf8729829721c9d21f223e/Lib/EmonLogger.php#L43

warpme, 2022-05-26:
ah - before reporting i made sure log file dir has proper rights!

borpin, 2022-05-26:
Can you put some debug code around this and see if you can determine what is going on?

Ah, the test is actually if the file does not exist and the `fopen` is used to create it (hence open and close).

The `FALSE` return means it cannot create the file.

Really needs a handler and an error to system log.

Untested...
```php
 if (!file_exists($this->logfile)) { 
     $fh = @fopen($this->logfile, "a");
     if (!$fh) {
        error_log("Log file could not be created");
     } else {
        @fclose($fh);
     }
 } 
```

warpme, 2022-05-27:
@borpin: your code works perfect!

borpin, 2022-05-27:
Ok, but you need to change the permissions so the log file can be created!

warpme, 2022-05-29:
Perfect. 
One issue less with php8.1 :-)
I see there is dedicated issue report related to php8.1: https://github.com/emoncms/emoncms/issues/1776
Looking on #1776 i went with https://github.com/thib66/emoncms/tree/php-8.1-fix---part1 repo and it works well for me on current archlinux.
  
So now I don't know what will be better:
1. close this issue with comment that there are other php8.1 issues with ref. to #1776 and #1777?
2. close this and reopen new one collecting all fixes needed in emoncms to run on current achlinux?

borpin, 2022-05-30:
It will get closed when this fix is incorporated.

8.1 is probably a bit more fussy.

TrystanLea, 2022-08-31:
I've merged @thib66's pull request which seems to work well for me here on 8.1, is the logfile error additional? I dont seem to be getting that I dont think?

warpme, 2022-12-08:
I just updated to v11.2.3 and all works perfectly!
Thx!
Closing this ticket.

closed 2022-12-08 by warpme