emoncms/emoncms #1807: My Electric app showing kWh for cost

techfixpros, 2022-06-07:
RPi 3B+
Raspbian Buster
emonCMS 11.0.9

The My Electric app shows the same numeric value for both kWh and cost on the app dashboard. The displayed MQTT value, and the value in the My Electric 2 is correct. I have deleted all app dashboards, and re-created, and the issue persists.
The parameters selected in both My Electric and My Electric 2 are use: power1pluspower2, and use_kWh: power1pluspower2_kWh for both apps.
[image]
[image]
[image]
[image]
[image]

techfixpros, 2022-06-08:
The mobile android app v2.3.0 also displays correctly.
[image]

cricalix, 2022-09-08:
Hah. I just filed this as https://github.com/emoncms/app/issues/174.
I'll hopefully get a chance to poke it tonight, and if I can fix it, I'll post a PR and we can hope it gets accepted.

cricalix, 2022-09-08:
@techfixpros Fix is attached to that other issue. If you're comfortable editing code on the Pi, it should be an easy fix to apply locally.

techfixpros, 2022-12-27:
Issue is resolved in main update.

closed 2022-12-27 by techfixpros