emoncms/emoncms #1811: Upgrading PHP version to 8.1

TrystanLea, 2022-08-31:
Notes on upgrading a php7.4 system to php8.1. Updates to emoncms still required, review in progress...

sudo add-apt-repository ppa:ondrej/php
sudo apt-get update
sudo apt install php8.1 
sudo apt install php8.1-curl php8.1-mbstring php8.1-xml php8.1-igbinary php8.1-dev php8.1-mysql

sudo pecl -d php_suffix=8.1 install redis
printf "extension=redis.so" | sudo tee /etc/php/8.1/mods-available/redis.ini 1>&2
sudo phpenmod redis

sudo a2dismod php7.4
sudo a2enmod php8.1
sudo service apache2 restart

mattheworres, 2023-03-06:
I'll tag a few of the PRs I've entered to help with the PHP 8.1+ migration: #1851 #1850

reedy, 2023-08-17:
OOI, do you actually still need to install redis from/via PECL?

I'm guessing there may have been a need at one point due to a "too old" version, but it's packaged in Debian (and as such, raspberry pi os derivatives etc), and in `ppa:ondrej/php` etc...

borpin, 2023-08-21:
> but it's packaged in Debian

It has been for a while, but yes, the packaged version was very old when first used IIRC. It is a bit of 'Ain't broke, don't fix' 😄

closed 2026-09-12 by TrystanLea