emoncms/emoncms #1819: Problem using Emoncms on windows

khriss75, 2022-10-30:
I know, it is no good idea to use Emoncms on Windows but I need to use this platform.
I have installed Wamp and Emoncms on my PC (Windows) with this step by step guide: https://github.com/emoncms/emoncms/blob/master/docs/WindowsInstall.md

I can see Emoncms login page, create an user and logged in.

The first problem, I see something like a pop up with this advise:
`wamp64\www\emoncms\locale.php on line 150
Call Stack
#	Time	Memory	Function	Location
1	0.0006	367608	{main}( )	...\index.php:0
2	0.0087	447072	set_emoncms_lang( $lang = 'en_GB' )	...\index.php:154
3	0.0087	447072	set_lang_by_user( $lang = 'en_GB' )	...\locale.php:163`

I can't solve this issue even if I set the language in user setting.
Moving on "Inputs" or "feed" I see only e "rotating cursor" no other information.

I think that I'm doing some error into the setting.ini, I can't really understand this as reported on Step by step guide: 

In the feed settings section set the datadir directories to relevant locations for windows e.g "C:\Users\Username\emoncmsdata\phpfina\".

On Windows '' must be escaped with another '' hence the '\'

This is what I insered in settings.ini file:
`phpfina[datadir] = 'C:\emoncmsdata\phpfina'
phptimeseries[datadir] = 'C:\emoncmsdata\phptimeseries'`

Can someone Windows user share with me the settings.ini file (deleting personal data)?

Moving on the "user" tab I see that error:
`
( ! ) Notice: Undefined index: total in C:\wamp64\www\emoncms\Modules\admin\Views\admin_main_view.php on line 133
--

1 | 0.0002 | 367608 | {main}( ) | ...\index.php:0
2 | 0.0249 | 448424 | controller( $controller_name = 'admin' ) | ...\index.php:242
3 | 0.0252 | 448952 | admin_controller( ) | ...\core.php:75
4 | 3.3447 | 467728 | view( $filepath = 'Modules/admin/Views/admin_main_view.php', $args = ['system' => ['date' => '2022-10-30 17:09:17 UTC', 'system' => 'Windows', 'kernel' => 'HP-PC', 'host' => 'NT', 'cpu_info' => FALSE, 'machine' => '', 'ip' => '::1', 'uptime' => '', 'http_server' => 'Apache/2.4.51 (Win64) PHP/7.4.26', 'php' => '7.4.26', 'zend' => '3.4.0', 'run_user' => 'nt authority\\system', 'run_group' => '', 'script_owner' => 'SYSTEM', 'db_server' => 'localhost', 'db_ip' => '127.0.0.1', 'db_version' => '5.7.36', 'db_stat' => 'Uptime: 7489 Threads: 1 Questions: 3948 Slow queries: 0 Opens: 192 Flush tables: 1 Open tables: 185 Queries per second avg: 0.527', 'db_date' => '2022-10-30 18:09:17 (UTC 01:00‌​)', 'redis_server' => 'localhost:6379', 'redis_ip' => '127.0.0.1', 'services' => [...], 'mqtt_server' => 'localhost', 'mqtt_ip' => '127.0.0.1', 'mqtt_port' => 1883, 'hostbyaddress' => FALSE, 'http_proto' => 'HTTP/1.1', 'http_mode' => 'CGI/1.1', 'http_port' => '80', 'php_modules' => [...], 'mem_info' => FALSE, 'partitions' => [...], 'component_summary' => 'Emoncms Core v11.2.3', 'git_branch' => '', 'git_URL' => '', 'git_describe' => ''], 'services' => [], 'redis_enabled' => FALSE, 'mqtt_enabled' => FALSE, 'emoncms_version' => '11.2.3', 'redis_info' => [], 'feed_settings' => ['engines_hidden' => [...], 'redisbuffer' => [...], 'phpfina' => [...], 'phptimeseries' => [...], 'cassandra' => [...], 'virtualfeed' => [...], 'mysqltimeseries' => [...], 'max_datapoints' => 8928, 'csv_decimal_places' => 2, 'csv_decimal_place_separator' => '.', 'csv_field_separator' => ',', 'csv_downloadlimit_mb' => 25], 'component_summary' => 'Emoncms Core v11.2.3', 'php_modules' => [0 => 'apache2handler v7.4.26', 1 => 'bcmath v7.4.26', 2 => 'bz2 v7.4.26', 3 => 'calendar v7.4.26', 4 => 'com_dotnet v7.4.26', 5 => 'Core v7.4.26', 6 => 'ctype v7.4.26', 7 => 'curl v7.4.26', 8 => 'date v7.4.26', 9 => 'dom v20031129', 10 => 'exif v7.4.26', 11 => 'fileinfo v7.4.26', 12 => 'filter v7.4.26', 13 => 'gd v7.4.26', 14 => 'gettext v7.4.26', 15 => 'gmp v7.4.26', 16 => 'hash v7.4.26', 17 => 'iconv v7.4.26', 18 => 'imap v7.4.26', 19 => 'intl v7.4.26', 20 => 'json v7.4.26', 21 => 'ldap v7.4.26', 22 => 'libxml v7.4.26', 23 => 'mbstring v7.4.26', 24 => 'mysqli v7.4.26', 25 => 'mysqlnd vmysqlnd 7.4.26', 26 => 'openssl v7.4.26', 27 => 'pcre v7.4.26', 28 => 'PDO v7.4.26', 29 => 'pdo_mysql v7.4.26', 30 => 'pdo_sqlite v7.4.26', 31 => 'Phar v7.4.26', 32 => 'readline v7.4.26', 33 => 'Reflection v7.4.26', 34 => 'session v7.4.26', 35 => 'SimpleXML v7.4.26', 36 => 'soap v7.4.26', 37 => 'sockets v7.4.26', 38 => 'SPL v7.4.26', 39 => 'sqlite3 v7.4.26', 40 => 'standard v7.4.26', 41 => 'tokenizer v7.4.26', 42 => 'xdebug v3.1.1', 43 => 'xml v7.4.26', 44 => 'xmlreader v7.4.26', 45 => 'xmlrpc v7.4.26', 46 => 'xmlwriter v7.4.26', 47 => 'xsl v7.4.26', 48 => 'Zend OPcache v7.4.26', 49 => 'zip v1.15.6', 50 => 'zlib v7.4.26'], 'mqtt_version' => 'n/a', 'rpi_info' => ['rev' => '', 'sn' => '', 'model' => '', 'emonpiRelease' => '', 'cputemp' => '', 'gputemp' => '', 'currentfs' => ''], 'ram_info' => ['used' => '', 'raw' => '', 'percent' => '', 'table' => '', 'swap' => ''], 'disk_info' => []] ) | ...\admin_controller.php:77
5 | 3.3470 | 486624 | include( 'C:\wamp64\www\emoncms\Modules\admin\Views\admin_main_view.php ) | ...\core.php:95
`

Thanks

Cris

firefox7518, 2022-10-31:
Hi

Is there a reason why you are using WAMP and not just IIS and PHP and MySQL installation? I use emonCMS and IIS since many years in production and I have more than 200 hits a second on it and the system is not even sweating...

borpin, 2022-10-31:
@khriss75 - this would be better asked on the forum https://community.openenergymonitor.org/latest, and if a specific issue identified, an Issue raised for that.

Please also format your message correctly :)

closed 2023-04-06 by TrystanLea