emoncms/emoncms #1821: editrealtime is not working

alexandrecuer, 2022-11-07:
[image]

alexandrecuer, 2022-11-07:
php var `$vis_version` not declared in [Modules/vis/visualisations/editrealtime.php](https://github.com/emoncms/emoncms/blob/master/Modules/vis/visualisations/editrealtime.php)

dont know why we need give a v variable to feed.js and vis.helper.js ?
```
<script language="javascript" type="text/javascript" src="<?php echo $path;?>Modules/feed/feed.js?v=<?php echo $vis_version; ?>"></script>
<script language="javascript" type="text/javascript" src="<?php echo $path;?>Lib/vis.helper.js?v=<?php echo $vis_version; ?>"></script>
```

borpin, 2023-01-04:
@TrystanLea bump 

https://community.openenergymonitor.org/t/datapoint-editing/22342/1

TrystanLea, 2023-04-06:
Yes agreed, data editing needs some work. Id like to integrate this in the graph module ideally and remove the original editrealtime/daily interfaces.