emoncms/emoncms #1823: Proposed solution to: access to dashboard from public (i.e. no user) gives wrong data representation respect to user access

And2IT, 2022-11-09:
Preamble: Emoncms installed on Raspberry, used in a timezone different from UTC. Only one user created in Emoncms, with correct timezone setting (for example: Europe/Rome).
What happens: when viewing a dashboard from logged user, date is correctly represented. When viewing the same dashboard from a public access (i.e. without logged user), date is wrong and also data are not correctly represented.

[image]

On the left, dashboard viewed after having logged in; on the right, the same dashboard viewed without having logged in.

I have resolved -temporarily- editing the file feed_controller.php, row 31:

  `      $user_timezone = 'Europe/Rome';`

instead of the original:

`        $user_timezone = 'UTC';`

I kindly ask to modify the master branch to apply the timezone of the emoncms user, also in case of public access to dashboards.

Thank you.

Jaddache, 2023-02-14:
> I am experiencing the same issue and the proposed change above has the graph displaying correctly now - although I now have a local change to the repository....
I also request the master branch to be modified as per @And2IT request above.

TrystanLea, 2023-04-06:
yes agreed a long standing issue, see: https://github.com/emoncms/emoncms/issues/1060 no immediate plan to work on this yet but it has been noted.

adelriop, 2023-12-11:
I also have this issue and don't know how to solve it. If I connect from my phone in my timezone to a public dashboard on emoncms.org it shows the right feed time stamp. However, I have an old kindle showing a screenshot of that dashboard. The screenshot is taken in a heroku app and it shows one hour less. I hope it get fixed someday.

And2IT, 2024-02-11:
> I also have this issue and don't know how to solve it. If I connect from my phone in my timezone to a public dashboard on emoncms.org it shows the right feed time stamp. However, I have an old [...]

You can temporary solve it with the proposed code modification as in my first post here.

And2IT, 2024-02-11:
@borpin 
Can you help me/us in pushing more effort to resolve the issue ?
Thanks !
Andrea

adelriop, 2024-02-12:
> You can temporary solve it with the proposed code modification as in my first post here.

I have my dashboard hosted by https://emoncms.org/ so I guess that change on the code is something that I cannot do and it would only be valid if I hosted my own emoncms, right? Thanks!

chaveiro, 2024-02-12:
Public dashboards use the browser location timezone. Of that is the case, no bug exist.

adelriop, 2024-02-12:
> Public dashboards use the browser location timezone. Of that is the case, no bug exist.
> Ensure device has the right region defined and use a supported browser.

As I mentioned on my post, I have a heroku app that takes the screenshot of the dashboard so show on a kindle device. I was not able to change the timezone of the heroku instance. The timezone of the heroku server that takes the screenshot is on a different timezone. I would like my dashboard to show the same time always -> https://emoncms.org/dashboard/view?id=62642

If you have any workaround, please let me know! Thanks.

chaveiro, 2024-02-12:
Pass the readkey of the user on the url so it gets logged in.

And2IT, 2024-02-13:
> Public dashboards use the browser location timezone. If that is the case, no bug exist.
> 
> Ensure device has the right region defined and use a supported browser.

As of my knowledge, having tested the locally installed emoncms on a Raspberry, the bug exists. In my install, the Raspberry (where Emoncms is installed) and the web browser are both in the same timezone (Europe/Rome) and it doen't work correctly when access is performed with no account. The only way to solve is hacking the code as told in first post.
@TrystanLea confirms the bug exists.

adelriop, 2024-02-13:
> Pass the readkey of the user on the url so it gets logged in.

I could not make it work. But I found a way to change the timezone of my heroku app and now it is working! Thanks!

TrystanLea, 2026-10-05:
Sorry for not engaging on this, I had been struggling with development capacity, something that is now much less of an issue what with claude etc, I've created a new issue as a basis for a larger piece of work on this here: https://github.com/emoncms/emoncms/issues/2015. Will get this implemented and shipped soon.

TrystanLea, 2026-10-05:
I will close this issue for now and continue on #2015

closed 2026-10-05 by TrystanLea