emoncms/emoncms #1825: INPUTS all going n/a NULL after recent update

njordan77, 2022-11-29:
Hi,
i'm a user of emoncms (image, not custom) for years now, syncing several SDM630 and SDM230. never any issues on a raspi3. great and stable like hell. but 2 days ago a did a FULL update stable, and now i have no more feeds/inputs anymore with data. I found a discussion where they mentioned the same issue recently and that a chmod to the logfile should help. did try this with 777 but even after several reboots, not sucessfully. i did a manual (terminal) install.sh but this did not change anything. emonhub is on 2.1.5.

I really hope this can be fixed easily if somebody knows whats the issue with this recent update.
This is one of the monents i hate myself for always trying to be up to date with modules.......so time consuming :-)

borpin, 2022-11-29:
Please close this as an issue and create a thread on the forum. https://community.openenergymonitor.org/latest

Do another full update. It should fix the issue.

njordan77, 2022-11-29:
I already did full update via GUI several times + reboots, nothing changed. anything else i need to think of.?

Maybe the log helps....but all worked fine until recent full update few days ago....

```
2022-11-29 19:17:10,035 ERROR    SDM230     Could not read all registers
2022-11-29 19:17:10,036 ERROR    SDM630     Could not read all registers
2022-11-29 19:17:10,037 INFO     SDM630     Connecting to Modbus device=/dev/ttyUSB1 baud=19200 parity=none datatype=float
2022-11-29 19:17:10,036 INFO     SDM230     Connecting to Modbus device=/dev/ttyUSB0 baud=9600 parity=none datatype=float
2022-11-29 19:17:10,036 ERROR    SDM630top2 Could not read all registers
2022-11-29 19:17:10,040 INFO     SDM630top2 Connecting to Modbus device=/dev/ttyUSB2 baud=19200 parity=none datatype=float
2022-11-29 19:17:10,047 ERROR    SDM630top1 Could not read all registers
2022-11-29 19:17:10,048 INFO     SDM630top1 Connecting to Modbus device=/dev/ttyUSB3 baud=19200 parity=none datatype=float
2022-11-29 19:17:20,079 ERROR    SDM630     Could not read all registers
2022-11-29 19:17:20,080 INFO     SDM630     Connecting to Modbus device=/dev/ttyUSB1 baud=19200 parity=none datatype=float
2022-11-29 19:17:20,081 ERROR    SDM630top1 Could not read all registers
2022-11-29 19:17:20,082 INFO     SDM630top1 Connecting to Modbus device=/dev/ttyUSB3 baud=19200 parity=none datatype=float
2022-11-29 19:17:20,083 ERROR    SDM230     Could not read all registers
2022-11-29 19:17:20,084 INFO     SDM230     Connecting to Modbus device=/dev/ttyUSB0 baud=9600 parity=none datatype=float
2022-11-29 19:17:20,085 ERROR    SDM630top2 Could not read all registers
2022-11-29 19:17:20,086 INFO     SDM630top2 Connecting to Modbus device=/dev/ttyUSB2 baud=19200 parity=none datatype=float
2022-11-29 19:17:30,029 ERROR    SDM630top1 Could not read all registers
2022-11-29 19:17:30,030 INFO     SDM630top1 Connecting to Modbus device=/dev/ttyUSB3 baud=19200 parity=none datatype=float
2022-11-29 19:17:30,035 ERROR    SDM630     Could not read all registers
2022-11-29 19:17:30,035 ERROR    SDM230     Could not read all registers
2022-11-29 19:17:30,037 INFO     SDM230     Connecting to Modbus device=/dev/ttyUSB0 baud=9600 parity=none datatype=float
2022-11-29 19:17:30,036 INFO     SDM630     Connecting to Modbus device=/dev/ttyUSB1 baud=19200 parity=none datatype=float
2022-11-29 19:17:30,036 ERROR    SDM630top2 Could not read all registers
2022-11-29 19:17:30,039 INFO     SDM630top2 Connecting to Modbus device=/dev/ttyUSB2 baud=19200 parity=none datatype=float
```

borpin, 2022-11-30:
Please close this as an issue and create a thread on the forum. https://community.openenergymonitor.org/latest

closed 2022-11-30 by njordan77