emoncms/emoncms #1828: Restore old Feed API help Page

borpin, 2022-12-17:
https://github.com/emoncms/emoncms/blob/d240d5133910cc16e0bdabb8adcd0420fd57f316/Modules/feed/Views/feedapi_view.php#L24

Please restore the old Feed API help page as it is a more complete list of the functions of the API.

borpin, 2023-04-07:
Certainly the authentication information.

It is difficult to work out what is missing, but here is a start

https://github.com/emoncms/emoncms/blob/d240d5133910cc16e0bdabb8adcd0420fd57f316/Modules/feed/Views/feedapi_view.php#L43

Assumes you only want a JSON response.

https://github.com/emoncms/emoncms/blob/d240d5133910cc16e0bdabb8adcd0420fd57f316/Modules/feed/Views/feedapi_view.php#L48

https://github.com/emoncms/emoncms/blob/d240d5133910cc16e0bdabb8adcd0420fd57f316/Modules/feed/Views/feedapi_view.php#L51

Delete **existing** I think it should be

[image]

I also don't think there is enough information. e.g.

[image]

What does this do? *Update Feed Feild* implies 'updating something that exists'. This seems to be creating something new.

In th **id** UI it says the name and then in the URL, the number. How about showing both in the UI box (and explaining that the number must be used)?

[image]

What do the bottom 5 parameters do?

TrystanLea, 2026-09-12:
Input, Feed, Device and Schedule API pages now all use a common format which includes the authentication info:

Example here from a locally hosted test account, keys not an issue

[image]

closed 2026-09-12 by TrystanLea