emoncms/emoncms #1831: Virtual feeds always zero

greentangerine, 2023-01-09:
Seems recent changes have rendered virtual feeds as permanently zero (although if you look at the graph/data, you can see it).

[image]
[image]
[image]
[image]

tatou100, 2023-01-10:
I'm also experiencing the same issue with 11.2.10.

TrystanLea, 2023-01-10:
Thanks, this is now fixed in 11.2.11

closed 2023-01-10 by TrystanLea