emoncms/emoncms #1834: TimeSeries data no longer stored after december update

gloorung, 2023-01-18:
Hi all,

I'm experiencing some issues with TimeSeries feeds since my update early december. More specifically, TimeSeries data is shown correctly in the Graph visualisation, but in other visualisations such as e.g. Zoom, Bar, ... recent data is no longer shown. 

I've narrowed down the issue the fact that the MYD file in the /var/opt directory is no longer updated. It's not entirely clear why this doesn't affect the Graph visualisation. Creating a new feed also seems to somehow temporarily fix the issue, but after some days, the feeds/file stops updating again. I have a second emoncms server running v 11.0.9 that doesn't have this issue. 

Any clue on how to fix this would be much appreciated :)

TrystanLea, 2023-01-18:
Hello @gloorung I think this is a bug in the zoom visualisation, can you try the latest emoncms master branch? I've just uploaded a fix.

gloorung, 2023-01-18:
Hi Tristan,

Thanks for the quick fix. Unfortunately, that didn't solve the problem. I've tried in different browsers after clearing the cache, but I still don't see recent data.

Is it normal that my .MYD files are not updated ?

gloorung, 2023-01-18:
seems I was too fast. In the meantime, the feedwriter service also stopped and after a restart, I can now see all data.

closed 2023-01-18 by gloorung