emoncms/emoncms #1845: Export broken for large datapoint and multi feeds

chaveiro, 2023-02-22:
From V10 to v11 the export mechanism in **feed_model** was changed and is now broken.

There are two issues created:

1 - Export datapoints are being limited, by the config to **max_datapoints** = 8928 for multi feeds even though it should not, and give errors like this on the CSV if exceeded the data points size:
  **Undefined array key 0 in   /Modules/feed/feed_model.php on line 822**

2 - Uses all memory. Previous streaming to the user agent was uses by dumping csv data to the browser as needed but now it tries to load everything in RAM before dumping to the browser as a file.

That poses an issue with large data ranges.  Eg. **Two months in 5min interval does not work!**

But increasing that limit in configuration allow the export to consume too much memory giving error like this.
**Fatal error: Allowed memory size of 134217728 bytes exhausted (tried to allocate 20480 bytes) in Modules/feed/engine/MysqlTimeSeries.php on line 282**

Please review this essential feature to emoncms customers that want to migrate to the recent versions due to php update and get a broken expory.

borpin, 2023-02-23:
> Please review this essential feature to emoncms customers that want to migrate to the recent versions due to php update and get a broken expory.

Use the USB import method if on a Pi or I simply do an `rsync` across hosts.

But yes, agree, needs to be fixed. Perhaps a check on size then a different method used. Not sure what the reason was for the change.

chaveiro, 2023-02-24:
Comment from an angry customer:
```
The way it worked before was ideal:
- select the feed or tag and click download
- select a year’s worth of data a 5 minute intervals 
- all the values in this feed were downloaded to a file name that included the date range and tag ID
If this can be repeated as a workflow that would be great.
```

@TrystanLea how was it possible for the export to become so broken?

TrystanLea, 2023-04-06:
Hey Chaveiro, to be honest, your post comes across quite rude and angry. My motivation to help is pretty much zero if you approach it this way. I get the impression from this post any many others that you are not happy with much of my recent work after v10 and honestly it is really tiring and makes me less likely to want to work on this.

I have tested this just now with PHPFina and it works fine streaming more than the max_datapoint limit without any trouble. Notice that the data point limit is only applied if the request mode is not CSV: https://github.com/emoncms/emoncms/blob/master/Modules/feed/feed_model.php#L611

Streaming of export data is also enabled in the MysqlTimeSeries implementation, see loading of the shared helper and setting of the csv header here: https://github.com/emoncms/emoncms/blob/master/Modules/feed/engine/MysqlTimeSeries.php#L253

The MysqlTimeSeries engine was updated to implemented all of this around December 2021.

If you want to reopen this or other issues, please try approaching in a more respectful way.

Solarbot, 2023-04-06:
Hi Trystan,

Firstly, thank you for all your work and time spent on the emoncms project - its great what you have implemented.  The story behind this issue is that Chaveiro has been working on my behalf.... I was using emoncms V9 running on the Azure servers which worked just great.  Microsoft then issued a notice that we had to make some technical changes before March this year which Chaveiro has been kind enough to help with as its not in my skillset - and as with alot of these things we got forced down a path with more implications than we would have guessed at the start!  I can't describe in detail what needed to happen but essentially we had to move from emoncms V9 to V10 and also a newer version of PHP along with other changes I think.  The upgrades and changes have all gone quite well thanks to Chaveiro's help but some of the core functions I use no longer work - may I describe my workflow in case you can suggest workflows or code changes to make this possible again?

Thanks for any help/pointers you might be able to provide, not sure what to do if I can't get this working again - yikes!

- I have, maybe 20, long term loggers sending data at 5 minute intervals to our emoncms on the azure servers.
- The data is stored here mainly to allow it to be pulled down remotely once or twice a year for further analysis.
- It is also used to warn when the logger goes off-line.
- To export I used to go to feeds, select the logger (which included all feeds for that device) and then clicked export.
- This generated a CSV file, with excel date stamps for the whole year, of all the sensor values on that device.
- The filename was ideal: startdate/time_enddate/time_device.csv
- V10 requires me to select each value on the device rather than being able to select all feeds for that device?
- The export function throws an error when I try to export data for a period greater than about 4 weeks?
- I think emailing is now not possible?
- V9 was also set to email me once a month with the CSV data from each device, as a backup.
- V9 sent an email warning me when a logger went offline for a period of time.
- We also had 'alarms' implemented so that if certain values were exceeded I would receive an email with this information.
- These functions don't seem to work now.
- Some/all of these things might be down to me not doing it right - apologies if this is the case :-)

chaveiro, 2023-04-06:
Hi Trystan, don't want to appear rude, apologies if that was the given fealing.
Just pointing out a problem to be fixed and the customer feedback, as usual.