emoncms/emoncms #1846: Link to "New: Debian build script" is dead

bonjour81, 2023-02-22:
On the emoncms main pagehere https://github.com/emoncms/emoncms     , the link [New: Debian build script](https://github.com/openenergymonitor/EmonScripts/tree/master/install/readme.md) is dead

borpin, 2023-02-23:
Fixed.

Any comments on using Debian appreciated as it is a generally unused distro. Personally I use Ubuntu as I know it works (20.04 *not* 22).

bonjour81, 2023-02-23:
Thanks for correction. It seems not yet visible but I can follow the pull request to find the link.
I'll keep you in touch about install. It will be on a fresh debian install on a wyse3040 hardware.
(I currently use emoncms on docker but for mqtt, last updates etc, I will move on a dedicated device)

borpin, 2023-02-26:
No it won't until it gets merged.

closed 2023-04-05 by TrystanLea