emoncms/emoncms #1848: Recommended way to run `emoncms` on non-Debian?

wimleers, 2023-02-23:
Hiya! 👋

I'm looking forward to adopting `emoncms` as soon as my heatpump arrives! 🤓😊 Most people in the Netherlands and Belgium (where I'm from) seem to be using custom solutions involving InfluxDB, RavenDB, Grafana, and so on. The barrier to entry is very high. `emoncms` seems like a better fit for mass adoption.

Plus, it runs on the ubiquitous LAMP stack, with which I've got tons of experience — I've been a Drupal contributor for >15 years 😊

So I was looking to already spin up an `emoncms` instance. I expected a simple `docker pull` or — better yet — downloading a `docker-compose.yml` file and running `docker compose up` but … https://hub.docker.com/r/openenergymonitor/emoncms/tags seems to not have been updated for a solid year! 😳

I'm working on open source software on a daily basis, so I _get_ that this can happen! But this surely makes getting started a lot more difficult 😅

So this leaves me wondering: **What _is_ the recommended way to run it?**

- I suspect the docs on this subject are thin because most users either rely on `emonPi` or https://github.com/openenergymonitor/EmonScripts/? Both assume Debian/Ubuntu.
- Perhaps simply nobody in the `emoncms` community is using `docker` to run `emoncms`?
- I need to run it on my M1 Mac Mini — hence Docker being preferred. I use `ddev` already to serve my personal site, and that works great. `emoncms-docker` unfortunately is AMD64-only (which works, but it's not quite ideal in terms of energy efficiency).

*I'd love to contribute*, but then I first have to get it up & running. Looking forward to pointers! 😊

chaveiro, 2023-02-24:
Hi, if you can help out getting this to work seamless in docker, please do. Should not be difficult.
I've been installing the traditional way, on a shared server or a VM with the underlying php / mysql installed.

borpin, 2023-02-26:
@wimleers - please close this and ask questions on https://community.openenergymonitor.org/

TrystanLea, 2023-04-05:
Hello  @wimleers thanks for the kind words and enthusiasm! I've updated the docker image :tada: and it now supports connecting to a MQTT broker (that must be running separately) to pull in data to emoncms.