emoncms/emoncms #1853: Virtual Feed Shortcomings

XtremeOwnageDotCom, 2023-04-06:
Migrated from #1660

I was working on creating a feed to display the amount of "self-produced" energy used at my house per day. The formula is pretty simple.

```
x = grid_energy_in_d: 9.4
y = load_energy_d: 31.1

p = 100 - ((x * 100) / y)
```

But, while trying to implement-

[image]

It became pretty apparent the ability to subtract the value from 100, doesn't appear to be easily possible. As well, you cannot use a virtual feed, inside of another virtual feed.

_Originally posted by @XtremeOwnageDotCom in https://github.com/emoncms/emoncms/issues/1660#issuecomment-1475474343_

A few potential solutions:

1. Allowing virtual feeds to reference other virtual feeds.
2. Allowing math operations to be reordered.

ie, instead of X - 1, allowing 1 - X.

TrystanLea, 2023-04-06:
Thanks @XtremeOwnageDotCom 

Reorganising the equation gives us a solution that can work with the existing implementation: 

    p = (((x*100)/y)-100)*-1

[image]

Not to say that we shouldn't explore the other solutions as I can see benefits to both.

XtremeOwnageDotCom, 2023-04-06:
I personally think both solutions have other differing use-cases which may benefit others as well.

I can see lots of intricate capabilities being introduced by allowing nested virtual feeds.  

For my example solution, I did previously resolve it in the way you described- But- still wanted to make sure to give a useful idea for future development.

TrystanLea, 2023-04-06:
Ok sure, I will take a look at nesting, ideally need something in the interface to detect recursion, that shouldn't be too difficult.

lo92fr, 2023-09-14:
Hello,

I take a look at point 1: "Allowing virtual feeds to reference other virtual feeds.".
It seems that it is already supported out of the box.

Just comment the following lines : 
        /*
        if (parseInt(this.feedlist[z].engine) == 7) { //input context and virtual feed and process writes to feed?
            continue; // Dont list virtual feed
        }
        */
in function 'showfeedoptions':function(processid){ in file process_ui.js do the trick.

I test basic behaviors like adding/substrating virtual feeds together, and it seems okay.
Perhaps just to verify as TrystanLea said if we need to add some recursion test to prevent loops.

Best regards,
      Laurent.

chaveiro, 2023-09-14:
Recursion is supervised, any feed missbehaving is disbles for editing.

lo92fr, 2023-09-14:
OK, perfect.
So is there something against the removal of these three lines so we can activate nexted feeds?

TrystanLea, 2025-07-16:
I've just tested removing this limitation on nesting virtual feeds in the latest version of emoncms and recursion in this case does cause a program crash due to full allocation of available memory if I for example create virtual feed vA that has a real source feed and then create vB that uses vA as a source feed (fine), but then if I try and add vB to vA and so end up creating an endless loop it fails, the "if ($steps > $total*2) {" check never gets activated as we're jumping from one process context to the next and so never going over that specific limit.

Did I break the functionality that automatically caught this type of recursion I wonder? 

Note: The if not virtual feed check has now been moved to https://github.com/emoncms/emoncms/blob/master/Modules/process/Views/process_ui.php#L190 in the latest version

closed 2026-10-05 by TrystanLea