emoncms/emoncms #1858: docker image for a complete emoncms installation and Home-Assistant add-on

alexandrecuer, 2023-08-17:
It is not an issue but rather a sort of overview of emoncms and docker

@TrystanLea : maybe I should move this somewhere else ?

I've explored the existing projects around making work emoncms in a docker container : 
- https://github.com/emoncms/emoncms-docker, which aims to have each part of emoncms running in separate containers, in the docker way. Anyway the project introduced the use of [supervisord](http://supervisord.org/) to run multiple services in the same container
- https://github.com/inverse/hassio-addon-emoncms/, an addon for [HomeAssistant](https://www.home-assistant.io/) using s6-overlay and nginx, where you have to run a mariadb container aside.

I may be wrong or not enough skilled with docker but but I haven't found anything that's totally self-contained and easy to handle.

So I've recently worked on a self-contained docker image for Emoncms using Alpine3.16, with modules graph, postprocess, sync, backup and dashboard, and based on the s6-overlay init system : https://github.com/just-containers/s6-overlay/

The result is a small multiarch image (aarch64, amd64, arm/v7) that can work as a Home-Assistant add-on but also as a standalone docker image if you are not in the Home-Assistant ecosystem

[image]

It is quite easy to operate as everything is inside : apache, redis, mariadb, mosquitto, the three main workers feedwriter, service-runner and emoncms-mqtt are operating as s6-services. So no configuration, you can change the mqtt username and password if you want, but that is the only thing you'll have to do, you don't have to run a specific container for mariadb.

It was not heavily tested but I checked in depth everything was working fine, including the modules side, so feel free to give a try if you have time.... 

the repo to include in the Home-Assistant add-on store, with everything necessary for the docker builds : 
https://github.com/Open-Building-Management/emoncms

[image]

if not using Home-Assistant, and want to test the image, without any data persistency :
```
docker pull alexjunk/emoncms:alpine3.16
docker run --rm -p 8081:80 -p 7883:1883 -it emoncms:alpine3.16
```
if you want persistent datas, create a folder named data on the host and start emoncms with a volume :
```
docker run --rm -p 8081:80 -p 7883:1883 -v /data:/data -it emoncms:alpine3.16
```
more details here : https://github.com/Open-Building-Management/emoncms/blob/main/DOCS.md

borpin, 2023-08-21:
@alexandrecuer - this is really good. Not had a chance to test it, but assuming it works it is magic.

@TrystanLea @glynhudson - I'd suggest we have an OEM Repo for this.

> * an addon for [HomeAssistant](https://www.home-assistant.io/) using s6-overlay and nginx, where you have to run a mariadb container aside.

Can I check, are you running the MariaDB inside the same container on HA?

glynhudson, 2023-08-31:
Sounds good, I've just tested it. I found I needed to add `alexjunk/` before the container name to run it i.e 

`docker run --rm -p 8081:80 -p 7883:1883 -it alexjunk/emoncms:alpine3.16`

It seems to work well, I tested connecting to the MQTT server and posting some data via MQTT. Nice work. 

It would be great to have this as a repo on the emoncms github, if I was to create a repo would you be happy to push it up? What do you think we should call the repo? Or alternatively we could just link to your repo? I would be happy with either, maybe it's easiest to add a note to our existing `emoncms-docker` repo to let users know that your al-in-one solution is available?

borpin, 2023-08-31:
I think it would be better as a repo on the OEM Organisation with @alexandrecuer as a contributor to it so he can update it directly.

alexandrecuer, 2023-09-12:
Thanks for the feedback

@glynhudson, I would say it depends on what you want to do in the future : 
1) using docker to install software on OEM hardware ? personally, I think this is the path to follow because it simplify maintenance.
2) just a dockerfile as developers are more and more asking for containers architectures ? 
3) maintaining a Home-Assistant add-on as it could be an interesting alternative in the HA ecosystem, as already mentionned by @borpin , where grafana is now the only monitoring solution ?

in case 1, the folder name in the OEM Organisation could be `containers` as you can be interested to have a repo with subfolders, each subfolder containing a specific Dockerfile, for example one for emoncms, one for emonhub and to produce all the images automatically with the github actions

in case 2, well, no need to maintain another repo from your side. Just link to my repo from the emoncms main README, in the install section : https://github.com/emoncms/emoncms#install

in case 3, if you feel it's good to leave the repo on the emoncms organisation, just use "addon-emoncms" for the repo name because it is what they use at home-assistant : https://github.com/hassio-addons/addon-grafana/

 If you prefer like @borpin to go for the OEM organisation, you can keep "emoncms" simply for the repo name....

github actions needs to be reconfigured with Repository secrets in order to send the builds to your docker hub (https://hub.docker.com/r/openenergymonitor/) 
[image]

There is also a nice option (which I have not tested yet) to publish to github packages (https://docs.github.com/en/actions/publishing-packages/publishing-docker-images#publishing-images-to-github-packages)

alexandrecuer, 2024-02-23:
created this page as a documentation : https://emoncms-docker.github.io
maybe I could add this link in the emoncms README ?
@TrystanLea : OK for that ?

TrystanLea, 2024-05-31:
Apologies for the delay, yes please :) 
I need to get back to testing your other docker image work that you emailed me about too, thanks for that!

alexandrecuer, 2024-05-31:
cf commit https://github.com/emoncms/emoncms/commit/f96380ff6e0efdac6644f74faab95c864f7658b4

closed 2024-05-31 by alexandrecuer