emoncms/emoncms #1870: Conflict between “apikey” in the url and hosted Emoncms servers

HLFCode, 2023-12-16:
This is an issue [raised on the forum](https://community.openenergymonitor.org/t/conflict-between-apikey-in-the-url-and-hosted-emoncms-servers/25101)
I host Emoncms on a Krystal Hosting server and we had a 4 day outage because of their Imunify360 rules.

It turns out that they block urls with the word “apikey” in because:
`The rule is detecting "apikey" from the URL, which is the name for a wordpress fakeplugin.
`
This “rule” took the whole site offline for IP addresses sending apikey in the url so I couldn’t even get in to manage the site…

They have (temporarily?) switched off the rule for my site but it will probably bite someone else.

Would it be possible to add an alternative word like “key” (leaving apikey so as not to break existing installations)

The rule includes other strings but they don’t look like they would cause a problem for Emoncms

`ModSecurity: Access denied with code 403, [Rule: 'REQUEST_URI' '@rx (wordpresscore|wp-zexit|wp-clearlineee|wp-resortpack|apikey|ioptimization|bqxtbuu|blnmrpb|wp-breeze|loftloader\.2\.4\.0|cve-2023-45124|root-file-manager|ph-file-manager|zer0day|file-manager-zeroday|phoenix_)'] [id "77350295"] [msg "IM360 WAF: Interaction with fake plugin||WPU:||T:LITESPEED||"] [severity "CRITICAL"] [tag "service_i360custom"] [tag "service_wp_plugin"]
`
Thanks for a great package
Mike

chaveiro, 2025-04-25:
It's not a emoncms issue, talk to your provider to consider exceptions.

TrystanLea, 2025-07-16:
Would providing apikey in the post body or http header circumvent this issue?

Use POST parameter (Recommended): "apikey=APIKEY"

Add the HTTP header: "Authorization: Bearer APIKEY"