emoncms/emoncms #1875: Docker hub image

devdems, 2024-03-29:
How is docker image on hub being built and where https://hub.docker.com/r/openenergymonitor/emoncms ?

alexandrecuer, 2024-03-31:
See https://github.com/emoncms/emoncms#docker-standalone-container

There are images here : 
https://hub.docker.com/r/alexjunk/emoncms

devdems, 2024-03-31:
I'm asking for the official one.

alexandrecuer, 2024-03-31:
There is no automated build for the "official" one

devdems, 2024-03-31:
Can we create one with github actions so we can keep the current official one up to date?

alexandrecuer, 2024-04-01:
Well, if you read the doc, the "official" image is experimental, not frequently updated. Currently, it is only for amd64 architecture, it is a not a full emoncms installation, it uses supervisord which is not the best option to dockerise a multi services application like emoncms.

That's why there is a suggested alternative using alpine linux and the s6 process supervision suite through [s6 overlay](https://github.com/just-containers/s6-overlay) which is more suitable to the emoncms architecture

devdems, 2024-04-04:
Even so. This official docker was built with some dockerfile. Let's put this dockerfile here, add github actions so it's regulary pushed to the hub even if it's not 100% to all docker standards. We can then fix it step by step to get it perfect.

The current image that is there is working just fine btw.

alexandrecuer, 2024-04-04:
Of course you can :-) Pull requests are welcome

See https://github.com/emoncms/emoncms-docker

devdems, 2024-04-07:
I would be glad to help, but i think it would be best to start with the dockefile that was used to create the current image that is on the hub. That is why i was asking for it in the fust post.

alexandrecuer, 2024-04-07:
@glynhudson : any ideas why the image on the hub was not produced with the dockerfile on the repo https://github.com/emoncms/emoncms-docker ?

closed 2025-07-16 by TrystanLea