emoncms/emoncms #1890: publish to mqtt from a http input process not working

Torsten099, 2024-06-28:
Hi,
I set up a new emoncms on a Raspberry Pi, and I copied my old database to it. 
Now I have a problem. Inputs which comes from a http post are not processed by "publish to mqtt" in the process list.

Only mqtt-inputs are published to the mqtt brooker.  

My question now, is there something changed over the time in emoncms that it is not more provided? In my old emoncms version
it was working. 

Nice greetings 
Torsten

borpin, 2024-06-29:
You would be better asking on the forum https://community.openenergymonitor.org/

Torsten099, 2024-06-30:
Ah, okay thx.

deancs, 2025-01-12:
I to have experienced this and it seems like a bug, it should'nt matter what type of API is used to input to a feed for the process list to executed. @Torsten099 did you have any feedback from openenergymonitor.org ?

TrystanLea, 2026-10-05:
Thanks for the report, and sorry for the slow reply.

Cause: the "Publish to MQTT" process only ran when the Mosquitto PHP extension was loaded in the PHP handling the request. MQTT inputs are processed by the emoncms_mqtt service under PHP CLI, where the extension is always loaded. HTTP inputs are processed by the web server's PHP (Apache or PHP-FPM), which often does not load it. The process then did nothing didnt log and error.

This is fixed in https://github.com/emoncms/emoncms/commit/364c2e3c2af989cd7926ea3e2a0164db00f391bf. The process now only needs MQTT to be enabled in settings and Redis to be available. Publishing is still handled by the emoncms_mqtt service.

closed 2026-10-05 by TrystanLea