emoncms/emoncms #1898: Demand shaper and Sync modules not working on Proxmox install

spike3312, 2024-10-04:
Hi all,

I have used proxmox for aproximatetly 2 years on a Lenovo timy factor computer for various things and thought I'd move my Node red and Emoncms instances from the Raspberry pi 4 to it (intention was before the SD card failed) oops too late, it locked up last weekend and has gone read only.

I installed Emoncms onto a container using the Ubuntu instuctions and it works fine other than demand shaper and sync modules will not start, looking at its logs I get the following.

`Oct  3 05:37:09 Emoncms systemd[1]: Stopped Emoncms Demandshaper service.
Oct  3 05:37:09 Emoncms systemd[1]: Started Emoncms Demandshaper service.
Oct  3 05:37:09 Emoncms systemd[1077808]: demandshaper.service: Failed to determine user credentials: No such process
Oct  3 05:37:09 Emoncms systemd[1077808]: demandshaper.service: Failed at step USER spawning /usr/bin/php: No such process
Oct  3 05:37:09 Emoncms systemd[1]: demandshaper.service: Main process exited, code=exited, status=217/USER
Oct  3 05:37:09 Emoncms systemd[1]: demandshaper.service: Failed with result 'exit-code'.
Oct  3 05:37:10 Emoncms systemd[1]: emoncms_sync.service: Scheduled restart job, restart counter is at 1501.
Oct  3 05:37:10 Emoncms systemd[1]: Stopped emoncms_sync service.
Oct  3 05:37:10 Emoncms systemd[1]: Started emoncms_sync service.
Oct  3 05:37:10 Emoncms systemd[1077809]: emoncms_sync.service: Failed to determine user credentials: No such process
Oct  3 05:37:10 Emoncms systemd[1077809]: emoncms_sync.service: Failed at step USER spawning /usr/bin/php: No such process
Oct  3 05:37:10 Emoncms systemd[1]: emoncms_sync.service: Main process exited, code=exited, status=217/USER
Oct  3 05:37:10 Emoncms systemd[1]: emoncms_sync.service: Failed with result 'exit-code'.
Oct  3 05:38:09 Emoncms systemd[1]: demandshaper.service: Scheduled restart job, restart counter is at 1502.`

I did another install with it and I got the same result, is this a result of using it on root user within the contrainer?

Anyone else had this issue?

spike3312, 2024-10-04:
PS the idea is to use the sync module to get the data off the Raspberry pi which is still running, I cannot log into the web interface as it shows Redis in error due to the SD card being in read only mode which is making me think the sync module will not be able to get the data,.

Next best idea will be to clone the SD card onto a new one then transfer data to the new VM, has anyone had success with this and it booting OK?

Otherwise I can get the feed files and the mysql database off it and put that into the new installation which I did years ago when the card in my Raspberry pi 3 failed and I upgraded to a 4 at the same time and that seemed to work ok, although I imagine not ideal.

borpin, 2024-10-08:
@spike3312 - you would be better asking this on the OEM forum.

borpin, 2024-10-08:
> I installed Emoncms onto a container using the Ubuntu instuctions

Did you create the 'pi' user?

spike3312, 2024-10-08:
> Did you create the 'pi' user?

I had not, it didn't as nor did it occur to me as I was no longer running it on a Raspberry Pi!

However it has sorted it out a treat, many thanks.

spike3312, 2024-10-08:
Issue closed, many thanks @borpin

closed 2024-10-08 by spike3312

borpin, 2024-10-09:
> I had not, it didn't

I just checked the docs and I was sure that creating the user `pi` used to be part of that. Did creating the user `pi` solve it, or did you do something else?

spike3312, 2024-10-09:
Creating the pi user has solved the issue yes, although I thought I had followed the instructions carefully I missed the step where it says enter your Ubuntu username here which would explain why I had the issue in the first place!

borpin, 2024-10-10:
Ok, good. yes it creates a drop-in for the service file :)