emoncms/emoncms #1906: sh: 1: iostat: not found

reedy, 2024-11-07:
Seen in the apache error logs...

alexandrecuer, 2024-11-07:
Not sure but i guess you dont run emoncms on a raspberry and I think this is because the admin module is designed for the raspberry pi only. Outside that plateform, It works fine but returns errors....i also have these errors on the docker image..

reedy, 2024-11-07:
I'm running it on an emonpi, with the `emonpiRelease: emonSD-01Feb24` image

reedy, 2024-11-07:
Should be fixable by installing `sysstat`... but why it wasn't, but is called...

https://github.com/emoncms/development/blob/d9be580dddc010a4b491effc78c2b72c4d36d995/experimental/emon-py/docs/install.md?plain=1#L133 is the only systat reference I can see...

https://github.com/emoncms/emoncms/blob/982d3ca6ddbf336998465cf338822acb71bc6d29/Modules/admin/admin_model.php#L508

Code using `iostat` has been around a while, but the package not seemingly being installed anywhere

chaveiro, 2024-11-07:
EmomCMS was written to support many OS and platforms. 
If it's giving error on some, needs to be fixed to adapt the where it's running. Has always been like this.

reedy, 2024-11-07:
If the code is trying to use it... you'd think the images they build and provide it would include it...

reedy, 2024-11-07:
But similarly, doesn't seem to be documented it is needed..

You shouldn't have to look at web server logs to see what packages you may or may not need.

alexandrecuer, 2024-11-08:
> Should be fixable by installing `sysstat`... but why it wasn't, but is called... [...]

Using sysstat package seems indeed the solution. It is also available on alpine linux. I guess it was originaly included in raspbian by default but no more on raspi os...

https://pkgs.alpinelinux.org/packages?name=sysstat&branch=edge&repo=&arch=&origin=&flagged=&maintainer=

borpin, 2024-11-10:
> If the code is trying to use it... you'd think the images they build and provide it would include it...

It is highly likely it used to be included in the standard OS, but that changed. It happens. As a community project, we rely on the community noticing and offering fixes :).

If you have a fix, push a PR!

reedy, 2024-11-24:
Presumably it should be added to the install/update scripts somewhere over at https://github.com/openenergymonitor/EmonScripts

TrystanLea, 2025-07-18:
Commit to improve handling of this, agreed that this needs to be in the EmonScripts install will add that
https://github.com/emoncms/emoncms/commit/a3146a824ecee0d398ba90b71e8d7b5f1d7be189