emoncms/emoncms #1910: Home assistant integration error

nic313, 2024-12-15:
Home assistant is not able to integrate emoncms anymore:

> Registratore: homeassistant.components.sensor
Fonte: helpers/entity_platform.py:366
Integrazione: Sensore ([documentazione](https://www.home-assistant.io/integrations/sensor), [problemi](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
Prima occorrenza: 09:07:38 (1 occorrenze)
Ultimo accesso: 09:07:38

>Error while setting up emoncms platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 366, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/src/homeassistant/homeassistant/components/emoncms/sensor.py", line 320, in async_setup_entry
    elem["unit"],
    ~~~~^^^^^^^^
KeyError: 'unit'

alexandrecuer, 2024-12-15:
Could you please give more context, i guess you use last version of home assistant zccordkng to your log extract but what is your emoncms version ?

borpin, 2024-12-16:
> KeyError: 'unit'

@nic313 Have you set up the units in Emoncms for the Feeds you are trying to import?

If not, select a feed in Emoncms, click on 'Edit' and ensure that *Unit* has a value.

[image]

@alexandrecuer - I think this error needs to be better handled and we need to add something to the Docs that this must be filled as part of the prerequisites.

Possibly should have been a breaking change as it is now required and wasn't before.

alexandrecuer, 2024-12-16:
@borpin : i use emoncms with datasets including many feeds without units and the integration when querying those datas does not crash like that...

Now, the emoncms feed api always return a unit field even for the feeds which do not have any unit set...unit field has been introduced in the api in 2018 : https://github.com/emoncms/emoncms/commit/58a5ffd4eaad038219847b7d33fd94d5c3e81785
So versions without unit field in the api reponse should be prior to 9.9.2 
https://github.com/emoncms/emoncms/releases/tag/V9.9.2

alexandrecuer, 2024-12-16:
@borpin : indeed you are rigth the error needs to be better handled, and there is something not very friendly which happens when there is no units since i've removed the friendly name. Anyway it is not crashing

so I've proposed a small change https://github.com/home-assistant/core/pull/133358

nic313, 2024-12-16:
Thank you @borpin
Setting up the unit from emoncms solved my issue!

closed 2024-12-16 by nic313

alexandrecuer, 2024-12-16:
@nic313 : could you juts tell us what emoncms version you are running ?