emoncms/emoncms #1914: Unable to load the feed from db.

bkrajendra, 2024-12-25:
Everything was working perfect, after server restart, I am getting error and not able to see any feeds.
In inputs it shows:
[image]

and in feeds page, page is not loading up any feed.

Getting following error in emoncms.log

```
2024-12-25 17:54:29.-495|ERROR|index.php|json_encode - feed: Malformed UTF-8 characters, possibly incorrectly encoded
```

Emoncms V 11.3.0

TrystanLea, 2024-12-28:
Hello @bkrajendra 

Do you see any errors if you look at the input/list.json api end point?

bkrajendra, 2025-01-03:
not sure why this issue came but what I observed was setting feed unit to °C was creating issue. I replaced all of these symbol with C and it worked. This was working fine earlier.