emoncms/emoncms #1915: Can not get passed the log in screen

MR-Apple-uk, 2024-12-27:
Just installed the Home Assistant addon version alpine3.20_emoncms11.6.5

When I click on the open web uni I cannot register an account or use an default account which I one of the help guides suggested

alexandrecuer, 2024-12-28:
What do you have in the addon log ?

Crash-Evans, 2025-02-08:
Hi Alex,
Also seeing this behaviour. Here is the add on logs:
s6-rc: info: service legacy-services successfully started
s6-rc: info: service sql_ready: starting
2025-02-08 18:58:30 0 [Note] Starting MariaDB 10.11.10-MariaDB source revision 3d0fb150289716ca75cd64d62823cf715ee47646 server_uid K7NPtqUuq1eV27pBklRKQl4BxRI= as process 143
[Sat Feb 08 18:58:30.198293 2025] [ssl:warn] [pid 138:tid 138] AH01906: localhost:443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
[Sat Feb 08 18:58:30.201673 2025] [ssl:warn] [pid 138:tid 138] AH01906: localhost:443:0 server certificate is a CA certificate (BasicConstraints: CA == TRUE !?)
2025-02-08 18:58:30 0 [Note] InnoDB: Compressed tables use zlib 1.3.1
2025-02-08 18:58:30 0 [Note] InnoDB: Number of transaction pools: 1
2025-02-08 18:58:30 0 [Note] InnoDB: Using crc32 + pclmulqdq instructions
2025-02-08 18:58:30 0 [Note] mysqld: O_TMPFILE is not supported on /var/tmp (disabling future attempts)
[Sat Feb 08 18:58:30.217884 2025] [mpm_prefork:notice] [pid 138:tid 138] AH00163: Apache/2.4.62 (Unix) OpenSSL/3.3.2 PHP/8.3.15 configured -- resuming normal operations
[Sat Feb 08 18:58:30.218217 2025] [core:notice] [pid 138:tid 138] AH00094: Command line: '/usr/sbin/httpd -D FOREGROUND'
2025-02-08 18:58:30 0 [Note] InnoDB: Using Linux native AIO
2025-02-08 18:58:30 0 [Note] InnoDB: Initializing buffer pool, total size = 128.000MiB, chunk size = 2.000MiB
2025-02-08 18:58:30 0 [Note] InnoDB: Completed initialization of buffer pool
2025-02-08 18:58:30 0 [Note] InnoDB: File system buffers for log disabled (block size=512 bytes)
2025-02-08 18:58:30 0 [Note] InnoDB: End of log at LSN=46980
2025-02-08 18:58:30 0 [Note] InnoDB: 128 rollback segments are active.
2025-02-08 18:58:30 0 [Note] InnoDB: Setting file './ibtmp1' size to 12.000MiB. Physically writing the file full; Please wait ...
2025-02-08 18:58:30 0 [Note] InnoDB: File './ibtmp1' size is now 12.000MiB.
2025-02-08 18:58:30 0 [Note] InnoDB: log sequence number 46980; transaction id 14
2025-02-08 18:58:30 0 [Note] InnoDB: Loading buffer pool(s) from /data/emoncms/mysql/ib_buffer_pool
2025-02-08 18:58:30 0 [Note] Plugin 'FEEDBACK' is disabled.
2025-02-08 18:58:30 0 [Note] Server socket created on IP: '0.0.0.0'.
2025-02-08 18:58:30 0 [Note] Server socket created on IP: '::'.
2025-02-08 18:58:30 0 [Note] InnoDB: Buffer pool(s) load completed at 250208 18:58:30
2025-02-08 18:58:30 0 [Note] mysqld: ready for connections.
Version: '10.11.10-MariaDB'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Alpine Linux
mysql running, so workers can be started
Securing mysql, creating database emoncms and user emoncms
s6-rc: info: service sql_ready successfully started
s6-rc: info: service sync_upload: starting
s6-rc: info: service service-runner: starting
s6-rc: info: service feedwriter: starting
s6-rc: info: service emoncms_mqtt: starting
s6-rc: info: service feedwriter successfully started
s6-rc: info: service service-runner successfully started
s6-rc: info: service emoncms_mqtt successfully started
s6-rc: info: service sync_upload successfully started
["CREATE TABLE `app` (`id` int(11) NOT NULL auto_increment PRIMARY KEY, `app` varchar(255), `userid` int(11), `name` varchar(255), `public` tinyint(1) DEFAULT '0', `config` text) ENGINE=MYISAM","CREATE TABLE `app_config` (`userid` int(11), `data` text) ENGINE=MYISAM","CREATE TABLE `myheatpump_daily_stats` (`id` int(11), `timestamp` int(11), `combined_elec_kwh` float, `combined_heat_kwh` float, `combined_cop` float, `combined_data_length` float, `combined_elec_mean` float, `combined_heat_mean` float, `combined_flowT_mean` float, `combined_returnT_mean` float, `combined_outsideT_mean` float, `combined_roomT_mean` float, `combined_prc_carnot` float, `combined_cooling_kwh` float, `combined_starts` float, `combined_starts_per_hour` float, `running_elec_kwh` float, `running_heat_kwh` float, `running_cop` float, `running_data_length` float, `running_elec_mean` float, `running_heat_mean` float, `running_flowT_mean` float, `running_returnT_mean` float, `running_outsideT_mean` float, `running_roomT_mean` float, `running_prc_carnot` float, `space_elec_kwh` float, `space_heat_kwh` float, `space_cop` float, `space_data_length` float, `space_elec_mean` float, `space_heat_mean` float, `space_flowT_mean` float, `space_returnT_mean` float, `space_outsideT_mean` float, `space_roomT_mean` float, `space_prc_carnot` float, `water_elec_kwh` float, `water_heat_kwh` float, `water_cop` float, `water_data_length` float, `water_elec_mean` float, `water_heat_mean` float, `water_flowT_mean` float, `water_returnT_mean` float, `water_outsideT_mean` float, `water_roomT_mean` float, `water_prc_carnot` float, `cooling_elec_kwh` float, `cooling_heat_kwh` float, `cooling_cop` float, `cooling_data_length` float, `cooling_elec_mean` float, `cooling_heat_mean` float, `cooling_flowT_mean` float, `cooling_returnT_mean` float, `cooling_outsideT_mean` float, `cooling_roomT_mean` float, `cooling_prc_carnot` float, `from_energy_feeds_elec_kwh` float, `from_energy_feeds_heat_kwh` float, `from_energy_feeds_cop` float, `quality_elec` float, `quality_heat` float, `quality_flowT` float, `quality_returnT` float, `quality_outsideT` float, `quality_roomT` float, `error_air` int(11), `error_air_kwh` float, `immersion_kwh` float, `unit_rate_agile` float, `unit_rate_cosy` float, `unit_rate_go` float) ENGINE=MYISAM","CREATE TABLE `dashboard` (`id` int(11) NOT NULL auto_increment PRIMARY KEY, `userid` int(11), `c [... cut ...] ication_key` varchar(64) DEFAULT '', `uuid` varchar(36) DEFAULT '') ENGINE=MYISAM","CREATE TABLE `rememberme` (`userid` int(11), `token` varchar(40), `persistentToken` varchar(40), `expire` datetime) ENGINE=MYISAM","CREATE TABLE `multigraph` (`id` int(11) NOT NULL auto_increment PRIMARY KEY, `name` text, `userid` int(11), `feedlist` text) ENGINE=MYISAM"]\nError: could not load feeds
Sync all
02/08/2025 07:01:41 pm
- Nothing to uploadStarting service-runner
Connected to redis server
 (sleep 198s)

alexandrecuer, 2025-02-08:
@Crash-Evans : did this happened after you've configured the sync module ? I think sync module is not well working in https ?

By the way, which app has created the table myheatpump_daily_stats ? (I dont use the app module)

Crash-Evans, 2025-02-09:
> @Crash-Evans : What is your home assistant version ?
> Through the log extract, we can see the addon crashing and rebooting after you log in . I cannot explain why....
> You are using safari on a imac...just to see do you have same behaviour with another browser ?

Hi Alex,
I am running 2025.2.1.

It is showing a dark screen exactly the same as the OP and it is not responsive to any actions. 

I have tried on my iPhone with the iOS app and see the same behaviour. I am away from home atm, but will dig a bit deeper when I get home.

alexandrecuer, 2025-02-09:
OK, maybe try only in http on your local lan, not https

what was your install method for home assistant ? this one ?
https://www.home-assistant.io/installation/generic-x86-64#method-1-installing-haos-via-ubuntu-booting-from-a-usb-flash-drive

alexalumb, 2025-03-27:
adalah ketika

TrystanLea, 2025-07-16:
Shall I close this? Was this resolved?