emoncms/emoncms #1919: emons.org server appears to be down

SwiftRR, 2025-02-15:
This is the url to my system:
https://emoncms.org/app/view?name=viessmann_combined&readkey=d9fa0b89bb710d76148a53d979624a2f

This morning (15th February 2025) at 05.02 am, feeds stopped from the emails server (for me).
I know the time because data was slowing to the emoncms entities in home assistant.
e.g.

[image]

My internal temperature monitor still shows data:

[image]

My internal network shows a steady connection (hardwired) to my emoncms kit.
I have restarted the emails kit but with no difference.
This is the error message that I see on trying to open my emoncms link.

[image]

I get the same on Safari and in Chrome. I have closed down and restarted both browsers but with no luck.
The issue seems to be with the emons server

Rob

TrystanLea, 2025-02-15:
This appears to be a bt server DNS issue outside of our control, see: https://community.openenergymonitor.org/t/issue-with-emoncms-org-server-sat-15th/28005

SwiftRR, 2025-02-15:
Thanks. I have followed your link and have read through the comments.
I can confirm that I can access my system on emoncms.org using mobile but any data input ubruptly stops just after 5am.
So we await bt (and indirectly ee) to sort out the issue.

Rob

SwiftRR, 2025-02-15:
emoncms.org is now back up for me, with service having resumed just after 13:00.

Here is the screenshot. I do have a big gap. Will this be rescued or is it just lost?

[image]

Thanks

Rob

alexalumb, 2025-03-27:
adalah ketika

closed 2025-07-09 by TrystanLea