emoncms/emoncms #1921: EmonCMS MySolar, MyElectric2 - blurry graphs

kobuki, 2025-03-28:
I’m trying to update an old EmonCMS installation to 11.6.10 using the alexjunk/emoncms docker image. All is working fine, I could import old input, feeds, data, etc. However, the graphs of MySolar and MyElectric2 are blurry. It’s like I’d need glasses to look at them, it’s very inconvenient. The old version (9.8.3) is perfectly crisp. As are the rest of the visual tools - the Visualization or the Graphs tools don’t exhibit this problem (among others), they’re as fine as in the old version.

After some digging, I noticed that in the HTML source the FlotChart canvas has 2 different sizes. One in the CSS and one in the surrounding element. The CSS one is 0.6px larger. This discrepancy seems to cause the blurriness. I’ve read on the net that in JQuery 3 the precision of certain functions changed to return fractions as well. This caused the issue in Flot which has been fixed. I see that JQ and its modules are very old, from 2017. I think it needs an update or at least a small fix for the old version.

I’m attaching 2 images, one where the +0.6px extra size exists, and one where it doesn’t (I hand-edited the HTML for the second image in the browser console). Even at this small size, the difference is quite apparent. Tried atest Firefox and Chrome.

[image]
[image]
[image]

I'm also attaching a simple patch that fixes the issue. It can be applied on `jquery.flot.js` (not `jquery.flot.min.js`), usable in place of `jquery.flot.min.js`. I can create a PR if this patch is a viable approach.

[blurry-patch.diff.txt](https://github.com/user-attachments/files/19507906/blurry-patch.diff.txt)

alexandrecuer, 2025-04-11:
Just to mention that JQ version is 3 6.0 so yes old but not that much. It was updated in 2021 : 
https://github.com/emoncms/emoncms/commit/564f9f126a099161177a618525310ba5fa809831
https://github.com/emoncms/emoncms/pull/1569

 Indeed trying to update to a recent version (3.7.1) should be nice...Did you try something on that ? 

For flot yes it is really old but the flot project is not maintained

@TrystanLea : you would accept a patch on that or everything that is in `Lib/flot` should remain something like frozen....

kobuki, 2025-04-11:
I haven't tried updating from upstream, seeing that it's very old in OEM and very likely wouldn't be compatible or break something.

TrystanLea, 2026-10-05:
Latest emoncms is running Flot v5.1 which is definitely sharper.

closed 2026-10-05 by TrystanLea