emoncms/emoncms #1924: Data Cuts off from Tuesday June 10th on the UK Grid Forecast Map

Johnr24, 2025-06-10:
Hey, First of all great work this is an absolutely excellent project, I make use of your UK gridforecast graph by embeding it into my home assistant instance. but I noticed recently the data cuts off from nationgrid, I think maybe they have closed or ended some endpoints which need updating as I can't seem to get to the data site in the same way as before. 

here is a screenshot 
[image]

Let me know if I can provide anything else!

Johnr24, 2025-06-18:
> Is there any fix for this?? I used the site daily to plan my energy usage - it was SO useful.

I had a breif look, and it would appear that an admin of that instance needs to go in and update some of the endpoints, from a quick glance it would appear the change is now  https://api.neso.energy rather than https://data.nationalgrideso.com/data-groups/demand 

I could be wrong but I don't beleive these values are "hardcoded" in the source code

TrystanLea, 2025-06-25:
Hello all, apologies with the delay, I thought I had updated all the scripts with [api.neso.energy](https://api.neso.energy/) and that there was something about the api returning zeros, but actually missed the crucial script that was actually fetching the data for this. I've now fixed that and the data is coming through again

ctyldsley, 2025-06-25:
> Hello all, apologies with the delay, I thought I had updated all the scripts with [api.neso.energy](https://api.neso.energy/) and that there was something about the api returning zeros, but actually [...]

Don't apologise - appreciate your work and for resolving this. Demand forecast data looks abit off for the next 24hrs but then jumps up for Friday 27th so assuming it's just a temporary glitch with the data kicking back in. Thanks for your help!

Johnr24, 2025-06-27:
closing the issue as it's all fixed, 

thanks so much!